OpenKinect/libfreenect2 #1173: CUDA build fails with GCC 12

jcelerier, 2022-08-28:
Here's the error:
```
$ make
[  6%] Built target generate_resources_tool
[ 10%] Building NVCC (Device) object CMakeFiles/cuda_compile_1.dir/src/cuda_compile_1_generated_cuda_depth_packet_processor.cu.o
/usr/include/c++/12.2.0/bits/locale_facets_nonio.tcc: In member function ‘_InIter std::__cxx11::time_get<_CharT, _InIter>::get(iter_type, iter_type, std::ios_base&, std::ios_base::iostate&, tm*, const char_type*, const char_type*) const’:
/usr/include/c++/12.2.0/bits/locale_facets_nonio.tcc:1477:77: error: invalid type argument of unary ‘*’ (have ‘int’)
 1477 |       if ((void*)(this->*(&time_get::do_get)) == (void*)(&time_get::do_get))
      |                                                                             ^   
/home/jcelerier/projets/packages/aur/libfreenect2-git/src/libfreenect2/src/cuda_depth_packet_processor.cu: In member function ‘void libfreenect2::CudaDepthPacketProcessorImpl::fill_trig_table(const libfreenect2::protocol::P0TablesResponse*)’:
/home/jcelerier/projets/packages/aur/libfreenect2-git/src/libfreenect2/src/cuda_depth_packet_processor.cu:883:23: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  883 |       const uint16_t *it0 = &p0table->p0table0[r * 512];
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jcelerier/projets/packages/aur/libfreenect2-git/src/libfreenect2/src/cuda_depth_packet_processor.cu:884:23: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  884 |       const uint16_t *it1 = &p0table->p0table1[r * 512];
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jcelerier/projets/packages/aur/libfreenect2-git/src/libfreenect2/src/cuda_depth_packet_processor.cu:885:23: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  885 |       const uint16_t *it2 = &p0table->p0table2[r * 512];
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at cuda_compile_1_generated_cuda_depth_packet_processor.cu.o.Release.cmake:280 (message):
  Error generating file
  /home/jcelerier/projets/packages/aur/libfreenect2-git/src/build/CMakeFiles/cuda_compile_1.dir/src/./cuda_compile_1_generated_cuda_depth_packet_processor.cu.o

make[2]: *** [CMakeFiles/freenect2.dir/build.make:384: CMakeFiles/cuda_compile_1.dir/src/cuda_compile_1_generated_cuda_depth_packet_processor.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:145: CMakeFiles/freenect2.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
[1]    31390 exit 2     LANG=C make -j1
```

bugmaschine, 2023-01-06:
I have the same issue

gseward, 2023-01-19:
Use GCC 11 instead.

To install GCC 11 on arch linux:
    
    pacman -S gcc11

Then, remove any prior build directories and before running `cmake` again, set GCC 11 as your compiler:

    export CC=gcc-11; export CXX=g++-11

Then you should be able to build without issues.