OpenKinect/libfreenect2 #1185: Linux compile error: expected unqualified-id before numeric constant

parkerlreed, 2023-09-12:
Overview Description:

Version, Platform, and Hardware Bug Found:

1. `fd64c5d (HEAD -> master, tag: v0.2.1, origin/master, origin/HEAD) Update README.md`
2. `Linux rogally 6.5.0-1-git-13150-g535a265d7f0d #1 SMP PREEMPT_DYNAMIC Sun, 10 Sep 2023 15:32:28 +0000 x86_64 GNU/Linux`
3. `gcc 13.2.1-3
opencv 4.8.0-5`

Steps to Reproduce:

1.  Try to compile

Actual Results:

Fails to compile

Expected Results:

Compiles

Additional Information:

```
[ 80%] Building CXX object CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o
[ 83%] Building CXX object CMakeFiles/freenect2.dir/src/opencl_kde_depth_packet_processor.cpp.o
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/cpu_depth_packet_processor.cpp: In member function ‘virtual void libfreenect2::CpuDepthPacketProcessor::loadP0TablesFromCommandResponse(unsigned char*, size_t)’:
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/cpu_depth_packet_processor.cpp:844:53: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  844 |     flipHorizontal(Mat<uint16_t>(424, 512, p0table->p0table0), impl_->p0_table0);
      |                                            ~~~~~~~~~^~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/cpu_depth_packet_processor.cpp:845:53: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  845 |     flipHorizontal(Mat<uint16_t>(424, 512, p0table->p0table1), impl_->p0_table1);
      |                                            ~~~~~~~~~^~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/cpu_depth_packet_processor.cpp:846:53: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  846 |     flipHorizontal(Mat<uint16_t>(424, 512, p0table->p0table2), impl_->p0_table2);
      |                                            ~~~~~~~~~^~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/cpu_depth_packet_processor.cpp:850:42: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  850 |     Mat<uint16_t> p00(424, 512, p0table->p0table0);
      |                                 ~~~~~~~~~^~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/cpu_depth_packet_processor.cpp:852:38: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  852 |     Mat<uint16_t>(424, 512, p0table->p0table1).copyTo(impl_->p0_table1);
      |                             ~~~~~~~~~^~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/cpu_depth_packet_processor.cpp:853:38: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  853 |     Mat<uint16_t>(424, 512, p0table->p0table2).copyTo(impl_->p0_table2);
      |                             ~~~~~~~~~^~~~~~~~
In file included from /usr/include/CL/cl.h:20,
                 from /usr/include/CL/opencl.h:24,
                 from /home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/include/internal/CL/cl.hpp:175,
                 from /home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_depth_packet_processor.cpp:50:
/usr/include/CL/cl_version.h:22:104: note: ‘#pragma message: cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)’
   22 | #pragma message("cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)")
      |                                                                                                        ^
In file included from /usr/include/CL/cl.h:20,
                 from /usr/include/CL/opencl.h:24,
                 from /home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/include/internal/CL/cl.hpp:175,
                 from /home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_kde_depth_packet_processor.cpp:50:
/usr/include/CL/cl_version.h:22:104: note: ‘#pragma message: cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)’
   22 | #pragma message("cl_version.h: CL_TARGET_OPENCL_VERSION is not defined. Defaulting to 300 (OpenCL 3.0)")
      |                                                                                                        ^
In file included from /usr/include/CL/opencl.h:26:
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_depth_packet_processor.cpp: In constructor ‘libfreenect2::OpenCLDepthPacketProcessorImpl::OpenCLDepthPacketProcessorImpl(int)’:
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_depth_packet_processor.cpp:254:15: error: expected unqualified-id before numeric constant
  254 |     const int CL_ICDL_VERSION = 2;
      |               ^~~~~~~~~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_depth_packet_processor.cpp: In member function ‘bool libfreenect2::OpenCLDepthPacketProcessorImpl::fill_trig_table(const libfreenect2::protocol::P0TablesResponse*)’:
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_depth_packet_processor.cpp:670:29: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  670 |       const uint16_t *it0 = &p0table->p0table0[r * 512];
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_depth_packet_processor.cpp:671:29: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  671 |       const uint16_t *it1 = &p0table->p0table1[r * 512];
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_depth_packet_processor.cpp:672:29: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  672 |       const uint16_t *it2 = &p0table->p0table2[r * 512];
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/CL/opencl.h:26:
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_kde_depth_packet_processor.cpp: In constructor ‘libfreenect2::OpenCLKdeDepthPacketProcessorImpl::OpenCLKdeDepthPacketProcessorImpl(int)’:
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_kde_depth_packet_processor.cpp:262:15: error: expected unqualified-id before numeric constant
  262 |     const int CL_ICDL_VERSION = 2;
      |               ^~~~~~~~~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_kde_depth_packet_processor.cpp: In member function ‘bool libfreenect2::OpenCLKdeDepthPacketProcessorImpl::fill_trig_table(const libfreenect2::protocol::P0TablesResponse*)’:
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_kde_depth_packet_processor.cpp:713:29: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  713 |       const uint16_t *it0 = &p0table->p0table0[r * 512];
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_kde_depth_packet_processor.cpp:714:29: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned pointer value [-Waddress-of-packed-member]
  714 |       const uint16_t *it1 = &p0table->p0table1[r * 512];
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_kde_depth_packet_processor.cpp:715:29: warning: taking address of packed member of ‘libfreenect2::protocol::P0TablesResponse’ may result in an unaligned point [... cut ...]
  715 |       const uint16_t *it2 = &p0table->p0table2[r * 512];
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/freenect2.dir/build.make:368: CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/freenect2.dir/build.make:382: CMakeFiles/freenect2.dir/src/opencl_kde_depth_packet_processor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:146: CMakeFiles/freenect2.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

parkerlreed, 2023-09-12:
Tried with clang 16 as well. Same results.

``
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_depth_packet_processor.cpp:254:15: error: expected unqualified-id
    const int CL_ICDL_VERSION = 2;
              ^
/usr/include/CL/cl_ext.h:3182:61: note: expanded from macro 'CL_ICDL_VERSION'
#define CL_ICDL_VERSION                                     2
                                                            ^
/home/parker/.cache/paru/clone/libfreenect2-git/src/libfreenect2/src/opencl_kde_depth_packet_processor.cpp:262:15: error: expected unqualified-id
    const int CL_ICDL_VERSION = 2;
              ^
/usr/include/CL/cl_ext.h:3182:61: note: expanded from macro 'CL_ICDL_VERSION'
#define CL_ICDL_VERSION                                     2
                                                            ^
6 warnings and 1 error generated.
make[2]: *** [CMakeFiles/freenect2.dir/build.make:368: CMakeFiles/freenect2.dir/src/opencl_depth_packet_processor.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
6 warnings and 1 error generated.
make[2]: *** [CMakeFiles/freenect2.dir/build.make:382: CMakeFiles/freenect2.dir/src/opencl_kde_depth_packet_processor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:146: CMakeFiles/freenect2.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

parkerlreed, 2023-09-21:
Removing libfreenect2's define from both ogf the opencl processors lets this compile

OpenCL already defines in its own files, version 2. libfreenect cannot reuse the same variable name.

arix-dev, 2023-11-11:
How do you go about doing that?

arix-dev, 2023-11-11:
Never mind figured it out by commenting out const int CL_ICDL_VERSION = 2 in opencl_depth_packet_processor.cpp and opencl_kde_depth_packet_processor.cpp

parkerlreed, 2023-11-15:
> Never mind figured it out by commenting out const int CL_ICDL_VERSION = 2 in opencl_depth_packet_processor.cpp and opencl_kde_depth_packet_processor.cpp

Yep you got it! Glad I could help